nexmonster/nexmon_csi #12: wlan0

marioqwert9, 2021-05-28:
[image]

I install all step but with wpa_supplicant the wlan is deleted. So how to fix this problem?

GuoJingtao-1997, 2021-06-06:
I also have this problem and I think it is due to the removal of wpasupplicant package. Have you fixed this it?

salehinejad, 2021-06-07:
I have faced the same problem.

GuoJingtao-1997, 2021-06-09:
> I have faced the same problem.

I reinstalled the patch and the problem was solved, you may try it.

CDeLeon94, 2021-09-16:
@GuoJingtao-1997 
How did you reinstall the patch?
Did you simply re-run the `install.sh` script?

I tried re-rerunning `install.sh` with tmux per the instructions and did not get wlan0 back. 
It may be worth noting that wlan0 disappeared after rebooting the device.

GuoJingtao-1997, 2021-09-17:
@CDeLeon94 
I also tried to re-run the `install.sh` script and worth nothing. So I refreshed the raspberry pi os after backup my materials and then reinstalled the patch using this one: https://github.com/zeroby0/nexmon_csi/tree/pi-5.4.51-plus and then it came back. A little confused.

zeroby0, 2021-09-29:
Okay so for some reason Github is not showing me notifications when issues are created here. And I'm soo surprised to see that issues were created and a discussion has been going on :)

Please tag me with @zeroby0 if you need help :)

The error is probably because you forgot to set the WiFi country. By default, rfkill in raspbian blocks wifi is country is not set. That's the reason [#prepare-raspberry-pi](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51#prepare-raspberry-pi) makes a special mention of setting it and timezone :)

@GuoJingtao-1997 thank you so much for helping people and in other issues :)

zeroby0, 2021-09-29:
Re-running install.sh doesn't help, and the install.sh actually has the same code here and in @zeroby0 / nexmon_csi

After running install.sh, a kernel module with modified firmware is created, and is made the default module. So rebooting the pi should be the same as re-running install.sh

Speaking of, I wonder if I can publish the modified kernel module as a pre-built binary, so people can do very fast installs.

GuoJingtao-1997, 2021-09-30:
@zeroby0 It is my pleasure :) I hope my answer can help others :)