nexmonster/nexmon_csi #15: Decoder code for bcm4366c0.

SoheiItahara, 2021-08-15:
@zeroby0 
Hi. 
I would be very grateful if you make a python code for decoding CSI from  bcm4366c0 (in nexmon_csi/utils/python/).
Your python decode code for bcm43455c0 is very useful for me to extract CSI from Raspi4.
But, I want to get CSI from multi-antennas chips (bcm4366c0 in Asus RT-AC86U).
And I can not use matlab for monetary reason....
If you make python decode code for bcm4366c0, this will promote my research in CSI.
Thank you!!

Sohei Itahara.

zeroby0, 2021-08-15:
Sure thing! :D

Is it urgent or can I do it a week later?

SoheiItahara, 2021-08-16:
@zeroby0 
Thank you so much!
If it is not too much trouble, I want the decode code urgently.
Of course, if you are busy, a week later is also helpful for me.
Thanks!!

zeroby0, 2021-09-29:
Hey @SoheiItahara 

I started writing it, but couldn't complete it because of a few things in personal life.. Sorry :)

If you made a decoder, I'd love it if you could make a pull request to the feature/python branch :)

ywy809080, 2021-12-20:
Hey @SoheiItahara
Have you solved the problem of writing python decoding code for bcm4366c0?
I want to use Python to solve the drawing problem.

nabeelni, 2022-09-15:
any decoder for bcm4366c0?

zeroby0, 2022-09-18:
If you have a pcap recording from bcm4366c0, I might be able to write a decoder this week.

nisarnabeel, 2022-09-19:
sure i will send you

nisarnabeel, 2022-09-19:
i sent you on mail

zeroby0, 2022-09-23:
Heyo!

I've implemented a basic decoder for all chips supported by Nexmon_CSI: https://github.com/nexmonster/nexcsi

I haven't tested it thoroughly, but the CSI values seem to be correct. Please create an issue there if you find that CSI values differ from the matlab script, or if you find any other bugs.

It reads all the values in the CSI payload, but the Timestamps in the pcap frames are not read yet. I'll implement that later. The decoding for raspberrypi is also 2 times faster than before, at ~640k samples per second. 

I'm also planning to write two other tools. One that plots CSI values (in realtime as well), and one that captures CSI packets in a more efficient format. It would also forward the packets over network to the plotter. All three tools should work nicely together.

Please test! 🙏🏻

nisarnabeel, 2022-09-23:
Thanks a lot Aravind,
I will check it out.
By the way did u ever try to modify  regulations file to add some channels?

Thanks !

zeroby0, 2022-09-25:
I remember trying to add all possible channel combinations to it, but I never got around to completing and testing it.

I think the procedure is to look up the channel index on https://en.wikipedia.org/wiki/List_of_WLAN_channels, then choose your bandwidth and the control channel.

So if you want to listen in the 5490–5570 MHz interval, `CH80MHZ_CHSPEC(106, WL_CHANSPEC_CTL_SB_L),`, and `mcp -c 100/80 ...`. The `100` is the control side band we set using `WL_CHANSPEC_CTL_SB_L`.  (This is speculation, I haven't tested it).

Or you could just write the chanspec, as described here: https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/utils/makecsiparams/bcmwifi_channels.c#L60