nexmonster/nexmon_csi #24: wlan0 not found error

yujianyuanhaha, 2021-12-01:
greetings,

I once have successfully install and been able to collect CSI at summer. But when I want to setup some other device I notice the manual does not work for unknown reason. The issue is `wlan0 not found error ` mention before @zeroby0 

I tries three way the 5.4.51, 5.4.51+ and 4.19.97. But all of them has `wlan0 not found error ` after reboot. 

The way I tried but not work is try to used the old version like
```
...
git clone https://github.com/seemoo-lab/nexmon.git
git checkout 1ad6a82
...
git clone https://github.com/zeroby0/nexmon_csi.git
git checkout aa370c6
...
```
but it does not work either.

Best

Jet

zeroby0, 2021-12-02:
Heyo Jet!

Which kernel are you running on the Pi? Can you post your `uname -a`?

I'll setup a pi tomorrow or on Saturday to see if I'll get the same error

yujianyuanhaha, 2021-12-03:
hi @zeroby0, I am following 5.41-plus version [https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus]

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
```

And the loggoing .txt file are at [debug folder](https://drive.google.com/drive/u/1/folders/1rXzMdLzgmXYUw6q7NN0VCOAkV2q8nEKg)

zeroby0, 2021-12-06:
Soooo I installed Nexmon_csi on an SD card, and didn't get any errors. Did you set wifi country via raspi-config at the start?

Also, the `wlan0 not found error` is when you run nexutil, right? I've recorded my screen when I was running it. I'll email you the link.

yujianyuanhaha, 2021-12-06:
@zeroby0 I do set `raspi-config`, and while `wlan0 not found error` come from `ifconfig wlan0 up`

zeroby0, 2021-12-06:
I made pre-compiled binaries. They seem to work for me, wanna try? These are for the  [pi-5.4.51 branch](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51). The install process takes less than a minute now 😆 

### Prepare Raspberry Pi
* Burn [Raspbian Buster Lite 2020-08-20](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/) onto an empty SD card. You can use [Etcher](https://www.balena.io/etcher/).
* [Create an empty file called `ssh`](https://www.raspberrypi.org/documentation/remote-access/ssh/), without any extension, on the boot partition of the SD card.
* [SSH](https://www.raspberrypi.org/documentation/remote-access/ssh/) into the Pi via Ethernet.
* With `sudo raspi-config`, set WiFi Country to US, set your Time Zone, and then Expand File System.
* Reboot when asked to.

- `wget https://github.com/nexmonster/nexmon_csi/files/7662477/nexbin.tar.gz`
- `tar -xzvf nexbin.tar.gz`
- `cd nexbin`
- `sudo bash install.sh`

Precompiled binaries: [nexbin.tar.gz](https://github.com/nexmonster/nexmon_csi/files/7662477/nexbin.tar.gz)

yujianyuanhaha, 2021-12-07:
@zeroby0  brilliant work, thanks a lot.

Best

Jet

dingyiyi0226, 2021-12-17:
Hi @zeroby0,
I prepared the rpi and install by this [solution](https://github.com/nexmonster/nexmon_csi/issues/24#issuecomment-987006153)

when i execute `sudo bash install.sh`, I get the following error
`insmod: ERROR: could not insert module ./brcmfmac.ko: Invalid module format`
and wlan0 is still not found
[image]

How to handle this problem?
Thanks

mzakharo, 2021-12-17:
Ok, so latest rpi is supported now, stop using these extra solutions, go directly to https://github.com/seemoo-lab/nexmon_csi#bcm43455c0

mzakharo, 2021-12-17:
just make sure to install `python-is-python2`

zeroby0, 2021-12-17:
I haven't tested it yet, but I hope the 5.10 version on seemoo-lab/nexmon_csi is much more stable than the previous versions. I love having this repo, but it would be better for CSI research if there is one version that works perfectly. So please test the seemoo-lab/nexmon_csi repo if you're experiencing problems here. Let's slowly migrate people to the main repo.

I might package the 5.10 version's pre-compiled binaries though, just so people can install faster

mzakharo, 2021-12-17:
faster installation can lead to errors. slow and steady wins the race

haochenku, 2021-12-19:
> 

Thank you @mzakharo , this new link works quite well on my raspberry pi. Great work. BTW, do you know which function in brcmfmac_4.19.y-nexmon folder is responsible for relaying the UDP packet to host/kernal 5500 port? 

Thanks a lot.

zeroby0, 2021-12-20:
I think this error is because either the driver or the firmware isn't loaded.

You can check by running `lsmod | grep brcm`. If you don't see `brcmfmac` in the output, the firmware hasn't been loaded. If `brcmutil` isn't present, it's the driver.

Try running `modprobe brcmutil` to load the driver. To load the firmware, `insmod brcmfmac.ko` in `nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon` folder. Or `insmod brcmfmac.ko.orig` to load the original firmware.

That said, there is a `/lib/firmware/brcm/brcmfmac43455-sdio.bin` file, and I don't know what it does, but [it is overwritten by Nexmon_csi](https://github.com/nexmonster/nexmon_csi/blob/pi-5.4.51/Makefile#L319). So the roles of `brcmfmac.ko` and `brcmfmac43455-sdio.bin` aren't really clear to me.

zeroby0, 2021-12-20:
@dingyiyi0226 

> when i execute sudo bash install.sh, I get the following error at installing nexmon_csi step
> insmod: ERROR: could not insert module ./brcmfmac.ko: Invalid module format

What is your kernel version, and which Raspberry Pi are you using?

dingyiyi0226, 2021-12-21:
@zeroby0 
kernel version: 5.4.51, Raspberry Pi 4B

[image]

zeroby0, 2021-12-22:
I wonder if it's because I'm compiling on a Pi 3B+. I don't have a free Pi 4B available to compile or test right now.

@yujianyuanhaha did you run the binaries on a Pi 4B?

avi3shake, 2023-05-04:
@dingyiyi0226 did you manage to solve the issue? I am getting the similar issue

dingyiyi0226, 2023-05-04:
@avi3shake Yes. I followed the instruction on https://github.com/seemoo-lab/nexmon_csi#bcm43455c0 instead of using `install.sh`