nexmonster/nexmon_csi #28: Problem understanding converting bytes to complex umbers

mamdouhJ, 2022-02-06:
Hello, 👋 
I am having trouble understanding how do we convert from bytes to complex numbers in csiexplorer

in `interleaved.py`: we read the bytes like a buffer then using `np.frombuffer()` we convert these values to decimal values. The issue I'm having is making sense of the numbers I'm getting. for example, The first 4 bytes the buffer is reading are 
`0xdf 0xf9 0x7f 0xfb` and the decimal output after` np.frombuffer()` is -1569 -1153. How are we getting the decimal numbers exactly? 

Thank you for your help. 

For reference:  (https://github.com/nexmonster/nexmon_csi/blob/feature/python/utils/python/decoders/interleaved.py)

zeroby0, 2022-02-07:
Heyo! 👋🏻 

The bytes are read in reverse, so for data `0xdf 0xf9 0x7f 0xfb`, the actual 16bit integers are `0xf9df` and `0xfb7f`. And they are written in 2's compliment form. So, https://www.rapidtables.com/convert/number/hex-to-decimal.html?x=f9df, check the 'Decimal from signed 2's complement' section.

The rest of the process is reshaping the data into a matrix with data of each sample in one row, and then adding alternate rows of the matrix to create a matrix of complex numbers. Doing operations on matrices are super fast because the data and instructions can be copied to the cpu cache, and executed in parallel on all cores. There are also special cpu assembly instructions for matrix operations, so it's fast.

And then we do an `fftshift` to get the order of subcarriers right. CSI is an fft of the training part of the packet, and like other ffts, has the negative and positive frequencies in the wrong order. .. atleast that's the way I understood it, I forgot why exactly I put an fftshift there, but it made the output consistent with the seemoo-lab matlab script. Please let me know if you have a better understanding. :)

mamdouhJ, 2022-02-12:
Thanks a lot for your help I got it now

closed 2022-02-12 by mamdouhJ