nexmonster/nexmon_csi #29: Extrac RSSI from the latest csi explorer?

chixiangw, 2022-05-23:
hi @zeroby0 , I hope you are doing well!

I'm now hoping to extract RSSI and CSI of each wifi frame simultaneously. 

I read your documentation and code of nexmon_csi under both nexmonster and also seemoo-lab, and it seems to me that only this branch [5.4.51-plus](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus) supports that function (i.e., I can get the RSSI and CSI of each frame simultaneously), and can be decoded by [this old version of csi_explorer](https://github.com/zeroby0/csi-explorer)? 

And it seems that the [seemoo-lab 5.10 version](https://github.com/seemoo-lab/nexmon_csi) doesn't support it because this newer one is using a different data format, which doesn't contain RSSI in data packets? And also, it seems that the latest python-version csi explorer (in [this branch](https://github.com/nexmonster/nexmon_csi/tree/feature/python)) cannot read RSSI either?

Is my understanding correct? Is 5.4.51-plus the only way, for now, to extract RSSI and CSI of each wifi frame simultaneously? 

Thank you!

zeroby0, 2022-05-25:
Hey @chixiangw 

Yes that's correct

But the new data format doesn't seem to be too different, so it shouldn't be too hard to update the parse. Lemme take a look

chixiangw, 2022-05-25:
thank you so much!! have a great day!