nexmonster/nexmon_csi #30: Setting up Rx and Tx

Taichi11111, 2022-06-22:
Hello. - Hello.
I am interested in obtaining a CSI due to my research.
I currently have the firmware installed on an ASUSU RT-AC86U and was able to run /jffs/tcpdump -i eth6 dst port 5500 and get the pcap file.
But I am not sure where this pcap file is the CSI of the communication from where to where.

I have two ASUSU RT-AC86U and I would like to communicate between them to get the CSI.

Please help me.

t5670116, 2022-06-28:
Thank you for your very clear explanation. Thanks to you I was able to acquire the CSI data.

I want to get the CSI of a communication using two 4×4 ASUSU RT-AC86U. So I used mcp -c 36/80 -C 1×f -N 1×f -m mac to run CSI.

But when the value of core is 2 or 3, the value of CSI is zero.
Please let me know if there is a solution or if my method is incorrect.

Also, is it correct that core represents the receiving router's antenna and N represents the transmitting router's antenna?
[image]

t5670116, 2022-06-28:
> Don't use the -b 0x88 flag btw. It limits you to only listening to broadcast packets - which are frequently only 20 MHz and are not beamformed.

I want to send a ping from my PC to the ASUSU RT-AC86 router and intercept it by the RT-AC86 router with firmware installed.

In that case, can I use -b 0x88 to intercept only the ping communication?

zeroby0, 2022-06-28:
> In that case, can I use -b 0x88 to intercept only the ping communication?

There's more packet types in the 0x88 class, so it's not *only* ping packets.

Also, for some reason, my devices seem to send ping packets as data instead of control packets. 

> But when the value of core is 2 or 3, the value of CSI is zero.

I don't know why CSI of core 2 and 3 is always 0. Maybe rt-ac86u only has a dual core processor.

> Also, is it correct that core represents the receiving router's antenna and N represents the transmitting router's antenna?

I think it's the actual processor cores. And N is the antennas. Of the receiving device.

t5670116, 2022-06-29:
Thank you very much.

I am aware from the comments here that each core (Rx antenna) and each "spatial stream" (Tx antenna) and if you select a particular core, it means you are using the antenna associated with that core.

It appears that the rt-ac86u only has a dual core processor as you mentioned. In that case how would you use the 4 antennas on the receiving side of the rt-ac86u?

I want to output the CSI value of 4(Rx antennas) x 4(Tx antennas) x subcarriers

[image]

t5670116, 2022-07-11:
@zeroby0
Hello.

I have discovered my mistake. The CSI value is 0 when the value of the spatial stream is 2 or 3, not the core.

/usr/sbin/wl -i eth6 monitor 1
If I use tcpdum to get CSI without this command, I can get 4x4 values.

Do you have any idea why the CSI value is 0 when the value of the spatial stream (N) is 2 or 3 and how to solve this?

/usr/sbin/wl -i eth6 monitor 1
What is this command doing? Also, what does the value obtained using tcpdum without this command represent?