nexmonster/nexmon_csi #31: tmux error

patorrad, 2022-07-18:
Hi,

I am getting this error in the last step (the tmux command line) of your github installation guide for kernel 5.4.51-plus:

```
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.c:21:10: fatal error: brcmu_utils.h: No such file or directory                                                           
 #include <brcmu_utils.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.o] Error 1                                                                     
make[1]: *** [Makefile:1709: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon] Error 2
make: *** [Makefile:108: brcmfmac.ko] Error 2
```

Any tips on how to fix this?

zeroby0, 2022-07-27:
I don't think I can debug this without further details, but my hunch is the build ENV variables haven't been setup. Did you run `source setup_env.sh`?

zikunliu, 2022-10-02:
@patorrad Hi, I have the same issues, have you able to solve this problems?

zeroby0, 2022-10-02:
Did you do:?
``` bash
source setup_env.sh
make
```
`source setup_env.sh` has to be done in every new terminal session.

zikunliu, 2022-10-02:
Hi Aravind, thanks for reply, I did 
```
source setup_env.sh
make
```
before 
`make install-firmware`

patorrad, 2022-10-03:
I did not find a solution. I had sourced the env variables so that was not my problem. I am still super interested in a solution.

zeroby0, 2022-10-03:
oh.. I don't know why it happens yet then.

Both `defs.h` and `brcm_utils.h` exist in the [include directory](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus/brcmfmac_5.4.y-nexmon/include). And `make` knows to look in that directory because of this:

https://github.com/nexmonster/nexmon_csi/blob/bf0a2bec37976b1e3c2c64c3286003ffad1d8757/brcmfmac_5.4.y-nexmon/Makefile#L18-L21

The path to the includes dir could be wrong if `NEXMON_ROOT` is wrong or unset. And `NEXMON_ROOT` is set when sourcing `setup_env.sh`

https://github.com/seemoo-lab/nexmon/blob/4087b4e5a408e6e1c4e5cbf4fd098b74bf74cdfa/setup_env.sh#L11

That's about all the information I have. I recommend wiping the SD card and starting over again, that miraculously solves bugs sometimes and is probably the quickest potential way to get rid of the bug. I'll try to build 5.4.51-plus later this week and see if I run into it too.

zikunliu, 2022-10-03:
Thanks Aravind. I can build 5.4.51-plus with kernal 5.10 by including folder(brcmfmac_5.10.y-nexmon) in your 5.04.51-plus and it is successed. I guess I could ignore error with kernal 5.4 since it is outdated.