nexmonster/nexmon_csi #33: Issue with installation

wanghe97, 2022-10-20:
When I run the command wget https://raw.githubusercontent.com/zeroby0/nexmon_csi/pi-5.10.92/install.sh -O install.sh
I got ERROR 404: Not Found

zeroby0, 2022-10-21:
Oh right, the link is for repository on @zeroby0, not @nexmonster, i made a typo 

I recommend using nexmon_csi_bin for getting started curl -fsSL https://raw.githubusercontent.com/nexmonster/nexmon_csi_bin/main/install.sh | sudo bash
If you want to compile from source, change the zeroby0 to nexmonster and try again

wanghe97, 2022-10-21:
Thanks for your reply. I want to enable RSSI function, which need to change the csi_extractor.c and csi.ucode.bcm43455c0.7_45_189.patch file. Does it still possible to use nexmon_csi_bin?

zeroby0, 2022-10-21:
With 5.10.92, RSSI and FCTL are included :)

The new decoder can read rssi and fctl from pcap files generated with 5.10.92: https://github.com/nexmonster/nexcsi

wanghe97, 2022-10-21:
Hi, when I execute the tmux command, it comes with an error (about: cannot find pi-5.10.92). Then I try to re-execute the tmux command line. However, this time another error comes when running 'git clone https://github.com/seemoo-lab/nexmon.git': 
fetal: destination path 'nexmon' already exists and is not an empty directory.
Could you help me solve this issue?