nexmonster/nexmon_csi #39: Nexutil command not working

Sankalp1Singh, 2023-03-15:
Im using a Raspberry pi 4b with raspbian os 32bit system
The installation ran smoothly
I keep getting this error when I run the 
nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==

I keep getting this error
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
[image]
 Please respond asap

Sankalp1Singh, 2023-03-16:
> Try restarting the pi :D 

Hii I tried it and it ended up taking out the wlan0 interface 
I faced this issue multiple times so I reflashed the sd card and restarted the entire installation but this is where I keep getting stuck.
Please help me out

zeroby0, 2023-03-17:
I would need more information. Post the exact steps you took to install and include as much information as you can.

The most frequent reason for issues is that the install instructions aren't followed exactly. I recommend installing https://github.com/nexmonster/nexmon_csi_bin on a fresh installation while following the instructions exactly.

The error can still occur if the firmware crashes, and the fastest and easiest fix is to restart the pi.

Also check if you're using the right kernel version: `uname -a`

Sankalp1Singh, 2023-03-20:
This is the output log of the installation. Here the patch files are installed in 7_45_189, but according to the firmware I have, ie 5.4.51-v7l+ and I should be installing the patch in 7_45_206.

DISASSEMBLING UCODE^[[0m /home/pi/nexmon/firmwares/bcm43455c0/7_45_189/ucode.bin => gen/ucode.asm
^[[0;31m  PATCHING UCODE^[[0m src/csi.ucode.bcm43455c0.7_45_189.patch => src/csi.ucode.bcm43455c0.7_45_189.asm
^[[0;31m  ASSEMBLING UCODE^[[0m src/csi.ucode.bcm43455c0.7_45_189.asm => gen/ucode.bin

[image]

closed 2023-03-28 by Sankalp1Singh