nexmonster/nexmon_csi #40: Amplitude when using matlab code

yama1111, 2023-05-10:
Hello. To answer your question, I would need as much detailed information as possible.

You described that you have installed the system on an ASUS RT-AC86U and obtained CSI by referring to Nexmon_csi #2. You also use iperf3 for UDP communication between a PC and another ASUS RT-AC86U to obtain CSI.

Then, you are using csireader.m to process the pcap file obtained from these two PCs, which is available at https://github.com/seemoo-lab/nexmon_csi/tree/master/utils/matlab.

The CSI I obtained is the first and second image.
I am fussy about the data as shown in the third image.

[image]
[image]
[image]

zeroby0, 2023-05-10:
Some subcarriers have arbitrarily high values for some reason. You can set their value to 0 so that the rest of the CSI is not over shadowed.

This article lists the positions of null and pilot subcarriers which have huge values: https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html

If you like Python, you can use nexcsi (https://github.com/nexmonster/nexcsi, `pip install nexcsi`) and plot them.

```python
import numpy as np
import matplotlib.pyplot as plt

from nexcsi import decoder

device = "rtac86u" # nexus5, nexus6p, rtac86u
samples = decoder(device).read_pcap('path-to-pcap-file.pcap')

# Accessing CSI as type complex64
csi = decoder(device).unpack(samples['csi'], zero_nulls=True, zero_pilots=True)

f, (ax1, ax2) = plt.subplots(2, 1)
ax1.plot(np.abs(csi[0])) # Amplitude
ax2.plot(np.angle(csi[0])) # Phase

plt.show()
```

[image]

yama1111, 2023-05-17:
Thank you for your prompt reply.
And sorry for the delay in getting back to you from us.

The point that the designated sub-carrier will have an expensive atai
I understand that by setting the value to 0, the value will be clean.

But the high value of the CSI data I mentioned seems to be coming from a different subcarrier than the specified location.

Thank you in advance for your help.

zeroby0, 2023-05-17:
If it's at different CSI locations, it's valid CSI data.