nexmonster/nexmon_csi #41: I installed it, but the mcp or tcpdump command doesn't come up

rack570, 2023-05-18:
hello, i'm the new of nexmon csi extractor. i installed script "https://raw.githubusercontent.com/nexmonster/nexmon_csi_bin/main/install.sh". and i used "mcp(makecsiparams)" command, but pi said "command not found"..
what should i do for extracting csi data? i already installed extractor, but it's difficult to use.

zeroby0, 2023-05-19:
Did you use just `mcp` or `mcp(makecsiparams)`? The command is just `mcp`

rack570, 2023-05-19:
i installed all parts of getting started section. then, i used "mcp" or "nexutil" but result is "command not found". Is there anything else to add? I definitely installed it, but the command doesn't work.. I'm sorry. I'm majoring in architectural engineering, so I don't know much about related fields..