nexmonster/nexmon_csi #42: After use nexutil -m2, I can't use wifie

sangeun-jo, 2023-05-23:
I use 5.10.92 kernel on Rpi3 b+.

After I use nexutil with -m2 parameter, I cant use wifi. 
I guess rpi still on monitor mode. 
Because after reboot rpi, I didn't turn off monitor mode but I could capture CSI. 
I can see wifi icon successfully connected on toolbar. But I cant use Internet with wifi. 
if I enter in chrome brower and try to search, browser says "no Internet" 

I tried below.
1. nexutil -m0
2. reload driver 
3. reinstall wpasupplicant package and start wpa_supplicant service 

I want to turn off monitor mode and use wifi. 
What else I can do before rewirte SD card?

zeroby0, 2023-05-24:
Yes it's normal. Use ethernet to access the pi.

Formatting the card will work, but if you installed using https://github.com/nexmonster/nexmon_csi_bin, downloading https://github.com/nexmonster/nexmon_csi_bin/blob/main/uninstall.sh and running it will also restore original firmware.

All you really need to do is 
- replace /lib/firmware/brcm/brcmfmac43455-sdio.bin with original brcmfmac43455-sdio.bin 
- replace brcmfmac.ko with original brcmfmac.ko
- run `depmod -a`
- install and enable wpa_supplicant

zeroby0, 2023-05-25:
Copying the original firmware, running depmod, installing and enabling wpa_supplicant should immediately restore wifi functionality without needing a restart.

I haven't experimented with just pkill ing wpa_supplicant instead of uninstalling it. If it works for you, please let us know.

sangeun-jo, 2023-05-26:
I recoverd wifi.
```
cp "$PWD/original/brcmfmac43455-sdio.bin" "/lib/firmware/brcm/brcmfmac43455-sdio.bin" //It was really needed
cp "$PWD/original/brcmfmac.ko" "$(modinfo brcmfmac -n)"
depmod -a
systemctl enable --now wpa_supplicant
```
But I want to make it easy. Is there some way to load driver per command like below?
LD_PRELOAD=[libaray] [capture command] // this command will excute with patched nexmon driver 
[capture command] //this command will excute without nexmon driver. 

[Reference](https://github.com/seemoo-lab/nexmon/issues/320)

closed 2023-05-30 by sangeun-jo