nexmonster/nexmon_csi #43: I can't build brcmfmac.ko

sangeun-jo, 2023-05-30:
I need to build nexmon csi by myself to use customized bandwide and chennel. 
If I can capture with nexmon_csi_bin project, please tell me I don't need to do below task. 

I tried to build brcmfmac.ko, but I got
```
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.c:12:10: fatal error: defs.h: No such file or directory
 #include <defs.h>
          ^~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.c:21:10: fatal error: brcmu_utils.h: No such file or directory
 #include <brcmu_utils.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.o] Error 1
make[1]: *** [Makefile:1709: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon] Error 2
make: *** [Makefile:108: brcmfmac.ko] Error 2

```

I found  brcmu_utils.h, defs.h file at nexmon_csi/brcmfmac_5.4.y-nexmon/include/ folder. 

but nexmon_csi/brcmfmac_5.4.y-nexmon/Makefile dosen't include headerfiles 

```
ccflags-y += \
        -I$(NEXMON_ROOT)/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_5.4.y-nexmon   \
        -I$(NEXMON_ROOT)/patches/bcm43455c0/7_45_189/nexmon/brcmfmac_5.4.y-nexmon/include \
        -DDEBUG

```
So I modified this. 

```
ccflags-y += \
        -I$(NEXMON_ROOT)/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon   \
        -I$(NEXMON_ROOT)/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/include \
        -DDEBUG

```

I solved importing header problem, but I got  another error. 

```
ERROR: "__sanitizer_cov_trace_cmp4" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp2" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp8" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp1" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp2" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_pc" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp4" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp1" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
ERROR: "__sanitizer_cov_trace_switch" [/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/brcmfmac.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1622: modules] Error 2
make: *** [Makefile:108: brcmfmac.ko] Error 2
```

kernel version 5.4.51-v7+

zeroby0, 2023-05-30:
It seems like `source setup_env.sh` wasn't run or didn't work properly for some reason. 

> If I can capture with nexmon_csi_bin project, please tell me I don't need to do below task.

Yup you can capture with nexmon_csi_bin :D

Check if the custom channel & bandwidth you need already work with the binaries from nexmon_csi_bin. If it doesn't, then you can look into compiling nexmon yourself.

sangeun-jo, 2023-05-31:
It is sometimes worked but not stable. And I need to wait long time to capture packets by specific mac address. 
I can't scan AP and I guess rpi3 hasn't changed with monitor mode. I made new [issue](https://github.com/seemoo-lab/nexmon_csi/issues/298) at seemo lab github.

sangeun-jo, 2023-06-01:
I've never been successful in 5 channel 20 bandwidth.  The bcm43455c0 chip can only collect CSI on limited bandwidth and channels. (157, 36 channel, 80MHz)I tried to get hex code through the -k option, but it doesn't seem to be the exact value. Do you know anything about this?

closed 2023-06-01 by sangeun-jo