nexmonster/nexmon_csi #51: What's the fastest way to extract CSI data and send it to a Windows laptop?

ChoiSeu, 2023-09-07:
Hello. First of all, thanks to nexmon_csi, nexmon installation is really easy.

I am not good at English, so there may be strange sentences because I use a translator. I'm sorry and I ask for your understanding.

My environment is as follows.
1. First raspberry pi 4 (for nexmon) => hereinafter nexmon pi
2. 20MHz/7channel TP Link WiFi Router
3. Windows OS laptop (for receiving extracted packets from nexmon raspberry pie and inferential HAR by ML)
4. Second raspberry pi 4 (for making traffic with laptop by iperf) => hereinafter iperf pi

[image]

I set it up like the picture above and imitated this git for my environment on nexmon pi.
```
mcp -C 1 -N 1 -c 7/20 -m<Wi-Fi Router's MAC>
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -v<mcp result>
iw dev wlan0 interface add mon0 type monitor
ip link set mon0 up
```
then,
```
tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 100
```

The creates a valid output.pcap with 100 packets measured in the folder where tcpdump is enabled.

To make machine learning inference on a laptop, I sent a CSI packet measured on nexmon pi to the laptop using Socket communication. But I want to send this CSI data to my laptop and infer it in real time and use it as a keyboard input for my laptop. The Socket communication I implemented can only send up to 8 packets. A bigger problem than Socket is that tcpdump takes about 1.2 seconds, whether it captures 8 packets or 100 packets. 1.2 seconds is not suitable for the function I want to implement, so I looked for a way. There are so many things I can't understand because I'm a beginner.

I've explained my situation so far and now it's a real question.

**Q1.** Is there a problem with the environment I set up? If there is no problem with the environment, is it right to do the behavior to measure in my environment between Router and nexmon pi? I'm using it because it's where the behavior is learned by trying many things.

**Q2.** Without iperf, capture speed is noticeably reduced. However, disconnecting and connecting the iperf changes the aspect of the csi data. Do you know how to measure a constant form of packet at a high speed other than iperf?

**Q3.** This is the most important question and I really want to know how to send extracted packets for real-time inference. Among your repositories, is relay what I want? I checked my nexmon pi and my laptop's local ip to create and set up setup.sh , but I don't know how the captured csi data will be sent to my laptop afterwards. When I write tcpdump -i eth0 dst port 5500, the ip I set comes up, but I don't know which folder on my laptop stores the packets under what name. I tried to study tcpdump and ports really hard but I didn't understand. I really hope you can give me a tip...

As I mentioned at the beginning, I am not good at English, so it may have sounded rude because I wrote it with a translator. I'm so sorry if I did. Lastly, thank you for reading the long article and I hope you always have a good day.

zeroby0, 2023-09-07:
Yes https://github.com/nexmonster/relay is what you want :)

The CSI packets will be forwarded to your laptop, and you can use a program to listen to UDP packets on port 5500 on your laptop and analyse them. 

If you want to save them to file on your laptop, you can do 
```
tcpdump -i eth0 dst port 5500 -vv -w captured-csi.pcap -c 100
```

But saving via tcpdump and then analysing them takes some time, so maybe directly read them in your ML program?

I'm a little bit busy this week, I'll read the question again and post a detailed reply soon :)

closed 2023-09-07 by ChoiSeu

ChoiSeu, 2023-09-07:
Thank you so much for your quick reply despite your busy schedule.
I'll try more with relay repositories, waiting for your next reply. What I understand is that in my environment, if I run "tcpdump -i eth0 dst port 5500" on nexmon pi and use a program like wireshark on my laptop, I can see that the captured packets are coming through the udp port.
Of course, I can experiment with it myself, and it might be good to leave the results and questions. However, I am replying first because I want to thank you for your reply as soon.
I hope you finish your busy work well and everything you do goes well! I'll be waiting!

Thank you again :)