nexmonster/nexmon_csi #53: nexmon_csi installation for raspi 4 model B kernel 6.1.21-v8+

psusarla95, 2023-09-21:
Hi,

I am new to the nexmon CSI tool and want to install it on my raspi 4 model B with kernel 6.1.21-v8+. The current github installation steps are dependent on the kernel version. Can I know what needs to be changed from the instructions to support for the latest kernel version ? I tried to downgrade as well but raspi forums mentioned that kernel version 5.15 has major security fix and is almost a standard now in subsequent kernel versions. Do you still suggest me to downgrade to 5.10 kernel version ? Please let me know

zeroby0, 2023-09-21:
Heyo,

I don't know how to do that either. Maybe ask in the seemoo-lab repo? They're more knowledgeable than me.

Yes I'd recommend using 5.10. Nexmon_csi is a only a research tool to get CSI data with which you'll do more interesting stuff later on. In locked down research/lab environments, you don't really need to worry about the security flaws. Or atleast, the effort to port nexmon_csi is disproportionately greater than the security benefits in this context.

psusarla95, 2023-09-26:
Hi,

Thank you for the inputs. I am now finally able to install mexmon_csi with Raspbian bullseye Desktop OS version 2022-01-28 with kernel 5.10.92 version.

I am now able to perform tcpdump following your usage instructions as well from here: [Usage](https://github.com/nexmonster/nexmon_csi/tree/pi-5.10.92#usage)

In the instructions, it was mentioned that python script will be released to extract and analyze CSI information. Is there such python script available by any chance ? The utils folder in the repository only has the matlab script currently.

Thank you for the help with this setup.

zeroby0, 2023-09-26:
https://github.com/nexmonster/nexcsi

Yup it's available! Nexcsi can decode the .pcap files. 

Here's example ipython notebooks for plotting amplitude and spectrograms: [nexcsi-examples.zip](https://github.com/nexmonster/nexmon_csi/files/12729502/nexcsi-examples.zip)

psusarla95, 2023-10-15:
Hi,

thanks for sharing the nexcsi tool to decode the .pcap file. I have been playing with this code from last 2 weeks now.

I want to read the tcpdump output in real-time and connect them to Python animation. I tried to save fixed number of tcpdump packets each time to a .pcap file and read them using nexcsi over python interface. However, I cannot run both the scripts in parallel as .pcap file is being read and overwritten with new UDP packets in the same time, resulting in error.

1. Is there a way to append these tcpdump packets to the .pcap files ?
2.  or Is there a way to capture the tcpdump packets directly from stdout and send them to the python animation for real-time updates ?

Have you tried this before by any chance? Any help regarding this ?

Regards,
praneeth

zeroby0, 2023-10-17:
You can use https://github.com/nexmonster/relay to forward packets from RaspberryPi to your computer over network. On your computer, you can have a python script listening for packets from Pi, decode each packet, and plot it. 

However, nexmon_csi can generate 1000s of packets per seconds and I don't think you can decode and plot them all, even if you could, your screen would only display 60 frames per second. So you'd have to decide which frames to decode and display.

While I haven't written anything to decode individual nexmon_csi packets yet, it should be fairly easy if you decide to write it yourself.