nexmonster/nexmon_csi #56: Issue of not receiving packets during csi extraction

rack570, 2023-10-12:
Hi, it's been a while since I wrote to you!
I'm using your tool well, but I have a question.
I installed the 5.10.92 kernel version of the nexmon extractor on Raspi 4B through your 'nexmon_csi_bin'. Sometimes the extraction goes well until the end, and sometimes the extraction stops in the middle. I thought this phenomenon was due to the instability of Raspi. But when I bought a few new raspberry pies and experimented with them, I found this phenomenon in the new Raspi 4B. Is there any instability in the nexmon extractor program itself?
Of course, it was useful to extract learning data because it is more often extracted, but I have to prepare for a situation where data extraction stops in the middle to extract 24 hours of test data a day. The sampling rates are 25Hz, and I'm extracting them with 5 minutes of extraction and 5 minutes of term for fear of overheating the device. I wonder if you have this experience. Or do you know anything about any instability in nexmon extractor?

zeroby0, 2023-10-14:
Hi! Good to hear from you again!

Nexmon_csi itself has some instability, the driver crashes after some time. I was able to run it overnight, for about 16 hours, without crashes, but I've never tried running it for longer than that.

Maybe you can have two Pi's setup very close to each other, and collect data on both of them. And have a script running which detects nexmon_csi crash and restarts the Pi. When one pi has crashed and is restarting, use data from the other pi.

You can detect crashes by listening to nexmon crash messages in dmesg, or by running `nexutil -k` and checking if the value is a valid chanspec.

rack570, 2024-02-07:
@zeroby0  Hi, I still haven't solved this problem. I'm using your nexmon csi bin, so I'll ask you again because I think I should ask you rather than asking seemoo nexmon. The logs below are output from dmesg. There seems to be no problem, but have you never experienced this? As you can see from the picture, I ran a repetition script. Sometimes I could do repetitions for a very long time, other times I got 514 packets during tcpdump after the 14th csi data extraction like this picture and was stopped in this state. After arbitrarily ending this state, even if I set chanspec and MAC again, csi packets will never be extracted again. Only reboot is the solution. Is the code to reboot raspberry pi valid in such cases as you told me above? I don't know if this state is because the tcpdump command stopped or if it doesn't capture csi packets.
[image]
[1]+  Stopped                 ./nexmon_v3.sh
root@raspberrypi:/home/pi# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.92-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1514 SMP Mon Jan 17 17:38:03 GMT 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1864704 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059584
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:0F:07:15 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=c678f507-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000012400000-0x0000000016400000] (64MB)
[    0.000000] Memory: 7758576K/8245248K available (10240K kernel code, 1366K rwdata, 3196K rodata, 2048K init, 882K bss, 158992K reserved, 327680K cma-reserved, 7458816K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34360 entries in 101 pages
[    0.000000] ftrace: allocated 101 pages with 4 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=1
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000095] bcm2835: system timer (irq = 26)
[    0.000757] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000779] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000802] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000819] Switching to timer-based delay loop, resolution 18ns
[    0.001092] Console: colour dummy device 80x30
[    0.001848] printk: console [tty1] enabled
[    0.001914] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001968] pid_max: default: 32768 minimum: 301
[    0.002146] LSM: Security Framework initializing
[    0.002348] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[... cut ...]
[  953.446321] device wlan0 left promiscuous mode
[  993.663987] device wlan0 entered promiscuous mode
[ 1013.457120] device wlan0 left promiscuous mode
[ 1053.654373] device wlan0 entered promiscuous mode
[ 1073.447614] device wlan0 left promiscuous mode
[ 1113.711574] device wlan0 entered promiscuous mode
[ 1133.448362] device wlan0 left promiscuous mode
[ 1173.675448] device wlan0 entered promiscuous mode
[ 1193.450528] device wlan0 left promiscuous mode
[ 1233.684057] device wlan0 entered promiscuous mode

rlajm1203, 2025-05-21:
@zeroby0 

Hi I'm using Raspberry Pi 4 Model B, kernel 5.10.x for CSI data collection. When I collect and show CSI, it seems that packets with different bandwidths are collected. (I'm guessing, but I think 20Mhz is collected.) When I read the official documentation of nexmon, I know that I can define the bandwidth that can be collected with mcp command and nexutil command. Is it possible that packets with different bandwidths are collected even if I define the bandwidth?

zeroby0, 2025-05-22:
@rlajm1203 I think it's more likely a software bug in my code than in nexmon. Checkout nexcsi, it's faster and simpler. 

As for bandwidth, yes, if you select 80MHz, 20MHz packets will be in your data too. And if you're filtering for 20MHz, the 20MHz of the 80MHz packets in the air that falls into the channel you're monitoring are included.