nexmonster/nexmon_csi #57: Use nexmon_csi extrator to monitor people behavior

crownsu, 2023-11-05:
I read the paper "Human Activity Recognition Using CSI Information
with Nexmon" to know the nexmon_csi extrator can help to monitor human behavior instead tranditional CCTV.
I study the from "https://github.com/nexmonster/nexmon_csi/tree/pi-5.10.92" and successfully dump the csi frame on my Pi 4 according the indication. 
Here I have a question that if we want to analyze csi data to determine the human behavior , it;s necessay to make a continoue data transmittion and receiveing traffic on air . But once we turn the Pi4 wlan0 into monitor mode , it won't be able to link another wifi AP or other wifi devices. How can we create a traffic on air and make sure Pi4 csi collectied data is by the targeted area wifi signal ?
What's the correct connectivity and necessary related devices in a room where we want to monitor ?

Thanks
Crown

zeroby0, 2023-11-06:
I connect the Pi to a WiFi router with Ethernet, and connect my laptop to the router with WiFi. Then I run iperf3 server on the Pi and iperf3 client on my laptop to generate wireless traffic between my laptop and the router.

crownsu, 2023-11-07:
> I connect the Pi to a WiFi router with Ethernet, and connect my laptop to the router with WiFi. Then I run iperf3 server on the Pi and iperf3 client on my laptop to generate wireless traffic between my laptop and the router.

OK ! should I set the IP following nexmon source 10.10.10.10 and destination 255.255.255.255 in pi and laptop ? Or just use the IP in the same LAN mask to be workable ?

zeroby0, 2023-11-08:
You should use the IPs that the router displays on it's settings page

crownsu, 2023-12-05:
I link my pi to a wifi router via ehternet like you suggested and set an ip camera link to router via wireless then use my pi to watch the camera . Then following nexmon csi capturing method to create a 1,000 frames wifi data under *.pcap , finally I extract into *.csv file .   But it's strange that I can not find my camera MAC address from the table of CSV file . All the source MAC in this table are not anyone of the devices I used including router , ip camera wire and wireless MACs . I can't filter the source before I nevigate the CSI data variance ~    Is anything wrong ?  looking forward to hear some comments and thanks so much !

crownsu, 2023-12-06:
> You should use the IPs that the router displays on it's settings page

> You should use the IPs that the router displays on it's settings page

I link my pi to a wifi router via ehternet like you suggested and set an ip camera link to router via wireless then use my pi to watch the camera . Then following nexmon csi capturing method to create a 1,000 frames wifi data under *.pcap , finally I extract into *.csv file . But it's strange that I can not find my camera MAC address from the table of CSV file . All the source MAC in this table are not anyone of the devices I used including router , ip camera wire and wireless MACs . I can't filter the source before I nevigate the CSI data variance ~ Is anything wrong ? looking forward to hear some comments and thanks so much !