nexmonster/nexmon_csi #59: Strange Mac Addresses

Akshay94, 2023-11-17:
I installed this repo on 2 of my raspberry pi and created a pcap file. But when i analyze it, I am not able to see any of the Mac addresses of my devices, (phone, laptop, etc) in the udp data. I realize that there is a possibility that the mac addresses may belong to my neighbors. But i scanned for a minute or so and none of the entries were my phone or computer or my friend's computer.
Additionally I note that some of the mac addresses have similar phrases among them. 

[image]

After looking a little bit more, I found that one of the mac is the xfinitywifi that is open. But why is that so?

Akshay94, 2023-11-18:
@zeroby0

zeroby0, 2023-11-20:
They'll only show up if they are operating in the wireless channel your nexmon_csi is listening to :)

The common parts in the mac addresses of multiple devices is the OUI prefix. Devices from the same manufactures typically have the same OUI prefix

Akshay94, 2023-11-20:
that solves the problem, thanks @zeroby0

closed 2023-11-20 by Akshay94

Akshay94, 2023-11-20:
@zeroby0 what are the units of these CSI values are these power values? or intensity? or raw amplitude? of the emwave..

zeroby0, 2023-11-20:
CSI is a ratio, so I don't think it has any units or dimensions. If you're using https://github.com/nexmonster/nexcsi, running unpack will return the CSI as a complex number. You can get amplitude and phase of CSI from that.

Akshay94, 2023-11-20:
Ok, so the amplitude and phase is that of the em waves?
@zeroby0

zeroby0, 2023-11-21:
No, of the CSI.

Refer to "Mathematical Description" in https://en.wikipedia.org/wiki/Channel_state_information and the free your csi paper. CSI is the channel matrix.

Akshay94, 2023-11-21:
I see. Thanks
I have another query if you've got the time to answer. I have two raspi 4Bs setup with nexmon. I want to collect CSI data in close quarters, like sender and reciever half a meter apart.
Since the wlan0 of Raspi is blocked for CSI capturing, I cant route packets to it as it is not assigned an IP even. So that would mean that the CSI-capturing-Raspi is not the receiver? And that the receiver has to be some other device necessarily? And what would you suggest as the transmitter?
And overall, what would you suggest my experimental setup be for my situation describe above. which devices should i use for tx and rx? (I have 2 Raspis, a laptop, an iPhone and a router)
We can take this conversation elsewhere, if you feel so.
@zeroby0