nexmonster/nexmon_csi #67: The capture file appears to be damaged or corrupt. (vwr: Invalid data length 3485744 (runs past the end of the record))

maple-42, 2024-06-24:
environment
Ubuntu 22.04
ASUS RT-AC86U

I referred to the code listed and successfully captured it with tcpdump.

ssh admin@IP address "/jffs/tcpdump -i eth6 dst port 5500" > ~/csi_data.pcap

and loaded the saved file with wireshark, I get the following error.
I am having trouble solving the problem and would appreciate your help.

[image]

maple-42, 2024-06-25:
As for the tcpdump log, this is what it looks like, so you should be able to get it.

15:33:04.163366 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.166634 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.169911 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.173185 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.176385 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.179644 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.182920 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.186195 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.189472 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.192748 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.196023 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.199298 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.202491 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.205757 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.209033 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.212311 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.215725 00:00:77:6c:31:00 (oui Unknown) > 41:00:00:00:90:00 (oui Unknown) Null Information, send seq 0, rcv seq 0, Flags [Command], length 160
15:33:04.215732 00:00:77:6c:31:00 (oui Unknown) > 41:00:00:00:90:00 (oui Unknown) Null Information, send seq 0, rcv seq 0, Flags [Command], length 158
15:33:04.218763 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
15:33:04.222037 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042

maple-42, 2024-07-02:
Still no resolution.
Please let me know when you have time.
@zeroby0

zeroby0, 2024-07-02:
No idea. Try plugging in a usb into the router and saving the pcap into it directly.

Or use nexmonster/relay to forward the csi packets to your laptop and store them on the laptop with tcpdump

maple-42, 2024-07-04:
Okay, Thank you very much.

maple-42, 2024-07-10:
I connected the USB to the router and tried saving the pcap directly, but the saved file was an empty file with 0 bytes.
Is there a solution?

zeroby0, 2024-07-10:
No idea. Hard to debug without access to device.