nexmonster/nexmon_csi #73: How to Receive Complex Numbers in CSI Matrix?

Jaesuru, 2024-10-09:
I am fairly new to using this and trying to process WiFi CSI data for experimenting, but after running it with a Raspberry Pi-3b+ using the Nexmonster patch firmware, I can successfully capture packet files and export them onto another computer to process them.

However, the problem is that after viewing each of the .pcap files that I got earlier, the CSI matrices obtained (processed with CSIKit) came out to contain only double negative numbers for each cell. 

An example of the script I used to process it was:
`csi_matrix, no_frames, no_subcarriers = csitools.get_CSI(csi_data, metric="amplitude")
`

After researching for a bit, I was told that each of the cells should contain complex numbers instead of what I got, and I want to know what the problem is. Again, since I'm new to this, I am not sure if it was the capture process or preprocessing part that gave me these unexpected results. If possible, it would also be helpful to know what the general process when utilizing CSI data and how to properly apply it for use.

Thank you.

zeroby0, 2024-10-20:
Maybe this should be reported in the csikit repository, remember to post your .pcap file aswell. Also try https://github.com/nexmonster/nexcsi

Jaesuru, 2024-10-20:
> Maybe this should be reported in the csikit repository, remember to post your .pcap file aswell. Also try https://github.com/nexmonster/nexcsi

Thank you for the reply. I will try these suggestions.