nexmonster/nexmon_csi #74: how to check that data packets contains the CSI information?

Ahmad6564, 2024-10-13:
[image]
[image]
[image]

I'm analyzing the [output.pcap ](https://output.pcap/?form=MG0AV3)file using Wireshark, but I want to know how to identify packets containing CSI information.

zeroby0, 2024-10-20:
If it fits the packet structure described in the nexmon_csi repo, then it's likely a csi packet. You can use https://github.com/nexmonster/nexcsi or https://github.com/Gi-z/CSIKit to see and plot the actual CSI data

Ahmad6564, 2024-10-20:
Thanks Sir for sharing insightful information.
I will contact if I get any query related to project.☺️