nexmonster/nexmon_csi #76: Get RSSI and CSI from .pcap file.

YiwenWang-YOLO, 2024-11-15:
Hello @zeroby0 ,

Very excited to see your work, I too would like to extract RSS and CSI from `.pcap` file at the simultaneous, I am getting my data from AC86U router.
I used MATLAB to read it and the same worked on https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus/utils  but no RSSI was extracted.

I want to extract the RSSI but I don't know exactly which byte corresponds to the RSSI, so I have no way to return the RSS. From the python code you shared I see that the RSS is extracted, can you share the byte that corresponds to the RSS from the pcap file?

YiwenWang-YOLO, 2024-11-15:
Do the byte symbols extracted from the `.pcap` file need to be converted to get the RSSI?

zeroby0, 2024-11-24:
Not sure if the matlab script implements rssi reading, and if it's implemented for ac86u at all. 

https://github.com/nexmonster/nexcsi/blob/ef2c32e0405ad45d1a6a04a9e8a98f0eefd6fa4d/nexcsi/interleaved.py#L129

https://github.com/nexmonster/nexcsi/blob/ef2c32e0405ad45d1a6a04a9e8a98f0eefd6fa4d/nexcsi/floating.py#L131

is where RSSI is extracted in the python code. The exact bit location is in the readme in the main nexmon_csi repo

zeroby0, 2024-11-26:
Pcaps will be different for pi and ac86u, specifically, the csi is encoded differently. Yes, my pcaps are from pi, which has only 1 antenna, so only one rssi. But if you use a mimo system, you still have only 1 rssi per packet capture because CSI detected by different antennas comes on different packets. The spatial stream indicator bit can be used to identify that.

closed 2024-11-27 by YiwenWang-YOLO