nexmonster/nexmon_csi #77: nexutil Problem

kuku000, 2024-11-22:
Hi,
I'm using raspberrypi 3B+ and  run install.sh,
However when I using nexutil ,I got this error.
root@raspberrypi:/home/pi# nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

zeroby0, 2024-11-24:
Reboot usually fixes things for me. The errors are sometimes because the patched firmware has crashed.

kuku000, 2024-11-25:
After reboot and I got the same error:
root@raspberrypi:/home/pi# mcp -c 157/80 -C 1 -N 1
m+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
root@raspberrypi:/home/pi# nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

so I try to re-make install-firmware:
`root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi# make install-firmware

          ###########   ###########   ##########    ##########
         ############  ############  ############  ############
         ##            ##            ##   ##   ##  ##        ##
         ##            ##            ##   ##   ##  ##        ##
         ###########   ####  ######  ##   ##   ##  ##    ######
          ###########  ####  #       ##   ##   ##  ##    #    #
                   ##  ##    ######  ##   ##   ##  ##    #    #
                   ##  ##    #       ##   ##   ##  ##    #    #
         ############  ##### ######  ##   ##   ##  ##### ######
         ###########    ###########  ##   ##   ##   ##########

            S E C U R E   M O B I L E   N E T W O R K I N G

                               presents:

              # ###   ###  #   # # ###  ###   ###  # ###
              ##   # #   #  # #  ##   ##   # #   # ##   #
              #    # #####   #   #    #    # #   # #    #
              #    # #      # #  #    #    # #   # #    #
              #    #  #### #   # #    #    #  ###  #    #

                The C-based Firmware Patching Framework

                           !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.c:12:10: fatal error: defs.h: No such file or directory
 #include <defs.h>
          ^~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.c:21:10: fatal error: brcmu_utils.h: No such file or directory
 #include <brcmu_utils.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.o] Error 1
make[1]: *** [Makefile:1709: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon] Error 2
make: *** [Makefile:108: brcmfmac.ko] Error 2
`

zeroby0, 2024-11-25:
Did you source the setup_env.sh before running make?

Also, try https://github.com/nexmonster/nexmon_csi_bin on a freshly installed raspbian if nothing works

zeroby0, 2024-11-26:
If I remembered right, I compiled them with support, but it's been more than an year and I can't be sure

closed 2024-11-27 by kuku000