nexmonster/nexmon_csi #78: MAC address filtering (URGENT)

ayyp1, 2024-11-27:
Hi @zeroby0 and @jlinktu  we meet again , 

when I extract CSI data using tcpdump without using the -m parameters on  the mcp step it takes the CSI packet from all the other APs except my router even when it is close by and in same channel 

Then when i use -m parameter with my mac address like mcp ...... -m <mac_addr>
the tcpdump is stuck on got 0 

can you please help me

xaxm007, 2024-11-27:
Hi @ayyp,

I also encountered the same issue. When I used -m <mac_address> to only capture packets from my router in the same channel as raspberry pi, it was stuck on GOT 0. 

Sometimes it the GOT packets increase but most of the times it is stuck.
I would like your help as well @zeroby0

jlinktu, 2024-11-27:
As the extractor works for you without address filter it seems obvious that you are using the filtering mechanism incorrectly.
Please have a look at "What MAC addresses shall be passed to makecsiparams -m?" of [the FAQ](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#frequently-asked-questions).

ayyp1, 2024-11-27:
Thank you @jlinktu  but I found my blunder 
There was no network traffic through my router so the nexmon couldnt get the packets , when i pinged to the router to create traffic and then tested tcpdump it worked like a charm. 
Learning new things everyday

closed 2024-11-27 by ayyp1

xaxm007, 2024-11-27:
> As the extractor works for you without address filter it seems obvious that you are using the filtering mechanism incorrectly. Please have a look at "What MAC addresses shall be passed to [...]

Appreciate your quick response @jlinktu.

It turns out the packets I got when I used tcpdump were not from my router but other router in my area which were on the same channel. 

Similar to @ayyp1, since I used a router with no internet access there were no other devices  connected to the router to create network traffic. After I connected other laptops to the router and used `ping <router-mac-addr> -s 1000`, I was able to capture the CSI data of only my router.