nexmonster/nexmon_csi #81: Question on extracting metadata in the src code (csi_extractor.c)

niruth99, 2025-01-16:
Hey, I’ve been reviewing the source code and had a question about `csi_extractor.c` for the chip version `CHIP_VER_BCM4355c0`. In the code, you extract metadata (source MAC, sequence count, and frame control) from `csi[tones]`. I was wondering if you could clarify the structure of `csi[tones]`, specifically the offsets to the different data variables, as I couldn’t find a definition for it.

Additionally, I was looking at your` pi-5.4.51-plus branch` where you extract additional metadata, such as the three MAC addresses from the MAC header. I noticed that in this branch, the offset used for extracting metadata from `csi[tones]` is different.

I had one more question— in  `csi_extractor.c` I’m assuming that `*p` refers to the entire packet from which the CSI information is being extracted. If that's the case, should I be able to extract additional metadata such as the source and destination IP addresses from it as well?

As I’m relatively new to C, it’s possible I may have missed something. Any feedback or clarification would be greatly appreciated.

zeroby0, 2025-01-20:
Might be helpful to refer to the Free your CSI paper: https://dl.acm.org/doi/10.1145/3349623.3355477

The `csi[tones]` buffer is populated in the firmware. However, I wrote those patches very long ago and don't remember the structure now, you'll have to experiment and find out. As for source and destination IP, I'm not sure if the DMAC core (refer to paper) even has access to IP layer details and only deals with MACs on the PHY layer. But if you can find how to access the IPs, sure, you can have them in the CSI packet if they fit in the packet header size (refer to paper). Might be easier to do IP lookups from MacIDs instead.

niruth99, 2025-01-21:
cheers mate, i will have a look though. if i come up with anything  ill send it through.