nexmonster/nexmon_csi #82: Random Packets

ayyp1, 2025-01-18:
Hi, 
I am currently using archer c6 ac1200 router as transmitter and I am trying to sense the csi packets using raspberry pi . Even though no device is connected to the router and there is no packet data is captured by tcpdump . Do you have any idea why ?

When I parse the data it shows the data is from my routers mac address 
Note: I have applied mac filtering in mcp and I am using channel 36 and 5ghz

zeroby0, 2025-01-20:
Router will periodically transmit beacons and other similar packets so devices nearby can detect it. If it doesn't transmit anything at all, your phone wouldn't be able to detect and connect to it, right? Typically their bandwidth is 20 Mhz.

closed 2025-01-20 by zeroby0

ayyp1, 2025-01-20:
I meant to say even though no device is sending ping signals I am able to collect csi data but in 2.4ghz this phenomenon is not seen why ?

zeroby0, 2025-01-21:
Maybe you're not using the right channel when using 2.4ghz. Collect with bandwidth 80 MHz  and mac filter in 2.4ghz and see if there's no packets at all.
Or maybe you're reusing the macaddress from 5ghz when collecting in 2.4ghz. The radios have different macids. Or there's a typo in the macid.

ayyp1, 2025-01-21:
so youre saying that in 5 ghz even though no device is pinging to the router or even though no device is connected to router ,  csi data can be collected using tcpdump ?
or am i wrong?

zeroby0, 2025-01-22:
Yup, if you place the router in outerspace with no devices within 3 lightyears and somehow power the router, it's still going to broadcast packets to announce it's presence. Pretty much all WiFi devices do this, afaik. It's done on both 5 Ghz and 2.4 GHz, and can be detected. 

But these broadcast packets use only 20 Mhz and sometimes a different modulation than normal datapackets. And their throughput is less than data packets, so they aren't sought after for CSI data collection.

https://en.wikipedia.org/wiki/Beacon_frame
https://en.wikipedia.org/wiki/802.11_frame_types
https://howiwifi.com/2020/07/13/802-11-frame-types-and-formats/

ayyp1, 2025-01-22:
Thanks man you're great